indigo-astronomy/indigo #625: Is indigo_ccd_pentax no longer included with 2.0.320

rockydev, 2025-03-01:
Updated to 2.0-320 on IndigoSky.

`
indigo@indigosky:~ $ indigo_server -v
21:05:38.552817 indigo_server: INDIGO server 2.0-320 Linux/armhf built on 2025-02-28T13:05:05+00:00 production
indigo@indigosky:~ $ indigo_server -i indigo_ccd_pentax
20:58:51.779286 indigo_server: Can't execute driver indigo_ccd_pentax (No such file or directory)
^Cindigo@indigosky:~ indigo_server indigo_ccd_pentax
20:59:00.272987 indigo_server: Driver indigo_ccd_pentax can't be loaded (indigo_ccd_pentax.so: cannot open shared object file: No such file or directory)
`

The website still lists it as available for 2.0-314 https://www.indigo-astronomy.org/hardware.html

I understand i can clone and build, but was wondering if this is the only driver i need .. is there an easier option to build just that driver? My reading of the docs didn't turn this up?

polakovic, 2025-03-02:
This driver is under development and doesn't really work yet :) So you can build it yourself, but it is not a part of a binary distribution.

rockydev, 2025-03-02:
Thanks, yeah ... I built it and it didn't work with K200D. The same with PTP or gphoto2 (obsolete) for some reason. I am trying to write Python adapter on [pktriggercord](https://pktriggercord.melda.info/) CLI as a Alpyca driver but something about the image generation and latency aren't working out. Hoping to understand the C code (not my best) to see if a CTYPE/CFFI would work better.

polakovic, 2025-03-03:
Yes, it is just a driver skeleton, camera detection and an attempt to read some camera properties, but no big success so far even if I purchased camera to do it :(