indigo-astronomy/indigo #627: Improve telescope limits with virtual horizon and meridian flip stop

Paolo97Gll, 2025-03-16:
Hi guys, we want to ask you a few questions about the limits implemented inside the mount agent.

- **How do the current limits work? Are they general and always valid, or are they to be set individually for each object?**
   - If the limits are general, could you clarify how they are meant to be used in practice?
   - If they are not general and should be set per object (which is, if we have understood correctly, what they are meant for), is there a way to dynamically adjust them based on the telescope's _goto_ coordinates?

Speaking of limits, we have two requests. We know that you are doing a lot of work right now and that new features are for now not accepted until you finish the Windows porting. Down the line, we would like to propose the following enhancements:

- **Virtual horizon**. It would be useful to include a feature for the virtual horizon. This would enable the telescope to avoid pointing below the local horizon, improving safety during operations and ease of use.

- **Meridian Flip Stop for Non-Automatic Flip Mounts.** Our mount, a Synscan EQ8 mount, does not support automatic meridian flip. It would be helpful to add an **option to trigger a stop or parking** when the telescope goes by a certain amount before or beyond the meridian (e.g. + or - 5 deg from the meridian). I know that AIN Imager displays the transit remaining time. However, this feature would ensure that the mount doesn't attempt to go beyond its physical limits, preventing any mechanical damage, and also if the operator forgets to stop the mount.

We have already implemented the last two points in an external Python service, but it would be better to have these limits implemented directly inside INDIGO.

polakovic, 2025-03-17:
Mount limits are global now, when either HA or time limit is hit, mount is parked (and everything else aborted). I agree, than some kind of virtual horizon is the final solution, it is on my TODO list for ages, but not implemented yet :(

Meridian flip is available already for some time, but works only with "new sequencer" as one of its part is implemented by script. On hardcoded side you can pause batch in imager agent if meridian is reached (and it can be parametrised how far beyond meridian you can track) and once it happens, sequencer will start automated meridian flip script - stops guiding, slews to current coordinates to force mount flip, executes sync & center with related solver agent and restores guiding. It is reported to work even if it is not perfect yet, e.g. it can't handle than flip happens during autofocus etc. Upcoming release will improve it  a little bit.

rumengb, 2025-03-17:
The new implementation of the meridian flip will fully handle the situation
described by Peter. It can also be used outside a sequence. You only need
to load the sequencer script (which implements the meridian flip logic) and
enable meridian flip by setting AGENT_IMAGER_PAUSE_AFTER_TRANSIT_FEATURE
then a normal batch will execute the meridian flip when needed, fully
synchronized  with exposures guiding etc.