indigo-astronomy/indigo #637: Issue: hard-coded UTC-UT1

Paolo97Gll, 2025-04-19:
Hi! By looking inside the code, I have noticed that the value of the UT1-UTC is hard-coded in the following files:

- [`indigo_libs/indigo/indigo_align.h`](https://github.com/indigo-astronomy/indigo/blob/63d74af79868d72f235318e77dc8c2e8d5090216/indigo_libs/indigo/indigo_align.h#L54)
- [`indigo_libs/indigo/indigo_driver.h`](https://github.com/indigo-astronomy/indigo/blob/63d74af79868d72f235318e77dc8c2e8d5090216/indigo_libs/indigo/indigo_driver.h#L53)
- [`indigo_libs/indigocat/indigocat_ss.c`](https://github.com/indigo-astronomy/indigo/blob/63d74af79868d72f235318e77dc8c2e8d5090216/indigo_libs/indigocat/indigocat_ss.c#L35)
- [`indigo_libs/indigocat/example/planets_test.c`](https://github.com/indigo-astronomy/indigo/blob/63d74af79868d72f235318e77dc8c2e8d5090216/indigo_libs/indigocat/example/planets_test.c#L7)

As you may know, this value is part of the [Earth Orientation Parameters (EOP) measured daily by the IERS](https://www.iers.org/IERS/EN/DataProducts/EarthOrientationData/eop.html). These values change daily due to the influence of the various bodies in the solar system and, for this reason, can't be predicted for a long time and shouldn't be hardcoded. For example, here is the variation of the UT1-UTC value:

[image]

(Just as a curiosity, note that the recent behavior is completely different from what we have seen in the past.)

The hardcoded value is -0.477677, while the current value is 0.034099.

I think that the ideal approach should be to download this value every day (e.g. with a timer), and leave a hard-coded value as a backup in case of no internet connection, updated at build time with the most recent one. I don't know how this approach alignes with your intentions.

Extracting this value is extremely simple since the syntax of the data published by the IERS is fixed. Data are published in the following formats: plaintext, CSV, JSON, and XML. For example, the following shell command extracts the most recent measured value of UT1-UTC, corresponding to the value for the current day:

```bash
curl -s 'https://datacenter.iers.org/products/eop/rapid/daily/csv/finals2000A.daily.csv' | awk -F ';' 'FNR == 91 {print $15}'
```

This command can be easily integrated into a Makefile and, in combination with the `sed` editor, can be used to update the hardcoded value at build time:

```bash
DELTA_UTC_UT1="$(curl -s 'https://datacenter.iers.org/products/eop/rapid/daily/csv/finals2000A.daily.csv' | awk -F ';' 'FNR == 90 {print $15}')"
sed -i "s%#define DELTA_UTC_UT1.*%#define DELTA_UTC_UT1 ($DELTA_UTC_UT1 / 86400.0)%g" indigo_libs/indigo/indigo_align.h
sed -i "s%#define DELTA_UTC_UT1.*%#define DELTA_UTC_UT1 ($DELTA_UTC_UT1 / 86400.0)%g" indigo_libs/indigo/indigo_driver.h
sed -i "s%#define DELTA_UTC_UT1.*%#define DELTA_UTC_UT1 ($DELTA_UTC_UT1 / 86400.0)%g" indigo_libs/indigocat/indigocat_ss.c
sed -i "s%double DELTA_UTC_UT1.*%double DELTA_UTC_UT1 = $DELTA_UTC_UT1 / 86400.0;%g" indigo_libs/indigocat/example/planets_test.c
```

rumengb, 2025-04-19:
I do not think +- 1s is a big deal this is why nobody bothered to update it :) But it is a good thing to keep it relatively up to date, so I created a slightly better script to update the value. However, this value will not be automatically updated on build. Thank you!

closed 2025-04-19 by rumengb

Paolo97Gll, 2025-05-13:
Everything depends on the desired accuracy: +- 1s is about +- 15" of pointing error, which I agree is negligible for most applications.

rumengb, 2025-05-13:
Well yes :)
This is why every new version of indigo will come with updated vaules...