indigo-astronomy/indigo #638: indigo_focuser_mypro2 does not connect to USB Gemini Focuser, but "-i indi_myfocuserpro2_focus" INDI driver works fine

tanvirh, 2025-04-28:
I get "MyFP2 AF did not respond" when I try to connect to my USB Gemini Focuser on /dev/ttyUSB0.  If I load up the INDI driver "indi_myfocuserpro2_focus" into Indigo on /dev/ttyUSB0 that works fine.  Turning on TRACE logging shows connection errors but nothing else.

tanvirh, 2025-04-28:
**Update**:
If I try to connect to either ttyUSB0 or ttyUSB1 with the indigo focuser driver it fails.  **But**, if I connect to my mount on ttyUSB1 and then try to connect again to ttyUSB0 with the indigo focuser driver it works.

rumengb, 2025-04-28:
please attach a trace log.

rumengb, 2025-05-22:
No trace log received and since it is very affordable i ordered a Gemini forcuser for development. The issue was that the focuser resets on first connect after powering it and it takes more time to respond to the first command and it was timing out. I fixed this issue and and added some optimizations in My focuser Pro 2 driver related to Gemini.

closed 2025-05-22 by rumengb