indigo-astronomy/indigo #650: alpaca driver reporting OSC camera Touptek2600C as mono camera

airborneastro, 2025-05-21:
Using N.I.N.A. connecting to a Touptek 2600C (OSC) color camera via indigo_server indigo_ccd_touptek indigo_agent_alpaca, (INDIGO Server 2.0-330 at raspberrypi), N.I.N.A. reports a monochrome camera and fails to show debayered previews. 
Checking with ASCOM Universal Device Conformance Checker results in the same:

22:48:03.362 ASCOM Universal Device Conformance Checker Version 4.1.0.34702, Build time: Di 17 Dezember 2024 13:02:48
22:48:03.372 
22:48:03.373 Operating system is Windows 10 (22H2), Processor is Intel/AMD 64bit, Application is 64bit.
22:48:03.373 
22:48:03.373 Alpaca device: Touptek ATR2600C (192.168.1.98:7624 Camera/2)
22:48:03.374 
22:48:03.378 CreateDevice                        INFO     Creating Alpaca device: IP address: 192.168.1.98, IP Port: 7624, Alpaca device number: 2
22:48:03.378 CreateDevice                        INFO     Alpaca device created OK
22:48:04.400 CreateDevice                        INFO     Successfully created driver
22:48:04.405 CreateDevice                        OK       Found a valid interface version: 3
22:48:04.406 CreateDevice                        OK       Driver instance created successfully
22:48:04.406 
22:48:04.407 Connect to device
22:48:05.916 Connected                           OK       Connected to device successfully using Connected = True
22:48:05.917 
22:48:05.927 Common Driver Methods
22:48:05.931 InterfaceVersion                    OK       3
22:48:05.934 Connected                           OK       True
22:48:05.939 Description                         OK       Touptek ATR2600C
22:48:05.941 DriverInfo                          OK       indigo_ccd_touptek
22:48:05.944 DriverVersion                       OK       2.0.0.38
22:48:05.949 Name                                OK       Touptek ATR2600C
.......
22:48:10.495 SensorType Read                     OK       Monochrome
22:48:10.505 BayerOffsetX Read                   OK       Sensor type is Monochrome and a NotImplementedException error was generated as expected
22:48:10.507 BayerOffsetY Read                   OK       Sensor type is Monochrome and a NotImplementedException error was generated as expected
....
Interestingly, "Sharpcap" 4.1 works with the same setup (not even restarting the rpi indigo_server) in the correct color mode: from the Sharpcap log:
 Info    21:13:50.936528 #1  Starting Preview on Touptek ATR2600C (#2) on 192.168.1.98 (INDIGO-Alpaca Bridge), 6224x4168, Bayer_RGGB, 16bits in Still mode.

airborneastro, 2025-05-22:
Just found that the indigo_ccd_touptek driver apparently does not support the bayered touptek camera, so please disregard.

isbeorn, 2025-05-22:
The property that defines whether a camera is a single-plane monochrome or a single-plane Bayer matrix sensor is determined solely by the following endpoint:
[](https://ascom-standards.org/newdocs/camera.html#Camera.SensorType)

As shown in the conformance check above, the driver incorrectly reports the camera as monochrome, which is not accurate for a One-Shot Color (OSC) camera. This misrepresentation leads to the incorrect image processing behavior in client applications.

rumengb, 2025-05-22:
Ah sorry it is my BAD! Apologies! it is not NINA's fault.

The problem is that INDIGO by design does not know in advance what is the sensor type so it can not provide it to ASCOM. And as a workaround we added a property in the Alapaca agent where the uses should enter the bayer pattern of the camera.

[image]

airborneastro, 2025-05-22:
Thank you both @rumengb and @isbeorn. @rumengb can you give me a hint how to access that parameter running indigo_server on rpi? I cannot remember to have seen that on the web server manager page. What application is shown in the above screenshot? Thanks, Karl

airborneastro, 2025-05-22:
Unfortunately, NINA then fails at exposure with Error: Unsupported CFA/Bayer pattern:-1. The camera reports with Sensor type: RGGB. as expected. I will file the trace log to the NINA discord @isbeorn as soon as my internet connection is back (on phone now).

rumengb, 2025-05-22:
What does the conformance test say? Is it OK with the image?

airborneastro, 2025-05-23:
The conformance test reports a color camera with RGGB and an ok status, so that seems to be ok. My internet provider was out for 16h so I have not yet discussed with @isbeorn why this workaround is not enough. It is a minor issue anyway as it only affects the screen preview. The fits files NINA saves when it reports a mono camera are ok and can be debayered externally with no issues.

airborneastro, 2025-05-23:
I just discovered that the same error in NINA occurs when the camera is directly connected to the Windows machine running NINA via USB, so I think the problem is elsewhere. The camera works fine on Kstars/Ekos using indigo on an Rpi5 (not using alpaca, as the camera is connected to the same Rpi5).