indigo-astronomy/indigo #669: Difficulty to plug onstep

Sylvain-BROCAS, 2025-08-31:
Hello,
Thank you for this amazing tool, Indigo is perfect for my project.
However, I have one problem : I can't link my onstep mount.

I found on the forum that onstep was supporting the LX200 protocole, nice! I tried... but here comes the problem : I can't achieve a connection between the onstep and indigo.

> I tried the onstep ascom driver with Nina, it worked. I tried with Ekos/Kstar + INDI server, it was tricky but it worked. In addition to that, I can communicate with the mount with LX200 messages with a simple serial command (arduino serial, picocom, minicom) and the serial port is well detected (by INDI and Indigo). That leads me to think that the mount is not causing the problem. I tried to use Onstep and generic in the LX200 parameters , and nothing.

 Here is how I proceed : I plug the mount on the pi, I boot the pi, then I go to the web control pannel, go to the LX200 parameters, set up and try to connect. It tries for a few seconds and fail then after a timeout.
 
 Here are the possible problems that could create this problem : 
 - I thought that it could be a bad baud rate setting, but I tried to force the baud rate to the good value, it didn't work. I then tried to change the baudrate in the onstep mount (originally 115200, thinking that the baudrate could be limited in indigo I set it at 9600). Did'nt work neither.
 - I though that maybe starting the pi with the mount plug could be a problem, but I tried to plug it afterward and no difference. Still doesn't want to connect.
 - It could be then a bad setting, but I can't see what I didn't tried to change
 - Or, it could be that onstepX is not supported (OnstepX >V10)
 -  - Maybe it could be that I don't use the right pi image

I'm using a raspberry pi 3 model B rev1.2, with this image : 2025-01-14-indigosky-armhf.zip (available here : https://www.indigo-astronomy.org/downloads.html)

I'm new with indigo, so I may have committed some mistakes or forgotten to do some specific actions.
I would appreciate a lot any help to find a way to connect my mount to indigo.

Thank you,
Sylvain

Sylvain-BROCAS, 2025-09-06:
Hello,
Thank you for your answer.

I'm sure that I selected the right serial port (/dev/ttyUSB0) because I was able to manually control the mount via a serial monitor. 
I'm planning to use indigo to control my mount and camera from a raspberry pi, and I want to use a custom webapp on my mobile to control it via the API. 

I used the web control pannel to select drivers and agents. I selected the LX200 driver and the mount agent, then clicked submit.
Then, in LX200 in the control pannel, I tried to connect LX200 mount (tried auto detect, Generic and Onstep with the right serial port, the one that works to controle the mount manually)

It tries for a few seconds and then turns red, can't connect.

I don't know if I can share any usefull log, or screenshot to help to find what I did wrong.
Clear sky,
Sylvain

polakovic, 2025-09-06:
If the port can't be opened by INDIGO server, it is definitely permission issue. But if there is some communication, it may be something else. Pls. try to set log level to "trace" and check the log (e.g. on http://indigosky.local:7624/log).

For the purpose you mentioned it is probably easier to use imager and mount agents and control them instead of control drivers directly. You will get many complicated high level features for free.

Sylvain-BROCAS, 2025-09-06:
I will try to get the logs tomorrow. 
For my purpose, I definitely need to connect to agents, this is planned. My problem occurs before trying to communicate to indigo, but really just to connect device (camera device is working btw).
But maybe I missed something. Am I supposed to connect from the driver or the mount agent or the mount driver ? Should I start LX200 server ?

rumengb, 2025-09-10:
Hi
The log you provided is partial. It starts after the communication attempt to the mount. Please enable Trace log and then try to connect to the mount.

rumengb, 2026-01-11:
> Hi guys,
> I finally found some time to investigate. The thing is that I had compiled
> my Onstep with baud rate = 115200 instead of 9600.
> Now that I rectified this detail, it works fine for Onstep mount
> connection.
>
> That's why my camera was working, the usb detection was working fine, but
> not the connection.

Thank you for clarifying :)